omaraflak/Arduino-Library #3: Changing baud rate

skgadi, 2018-02-21:
Hi,

This library is very very useful. I want to use this for a project which uses baud rate 115200. Is there any change to change the default baud rate from 9600 to 115200?

Thank you.

skgadi, 2018-02-22:
Thank you very much. It is working as expected.

OmarAflak, 2018-02-22:
I'm glad to help !

royseberg, 2018-08-14:
When I write:
Arduino arduino = new Arduino(Context, 115200);
the word Context is underlined in red and Expression expected pops up.
What's wrong?

OmarAflak, 2018-08-14:
@royseberg, `Arduino arduino = new Arduino(YourActivityName.this, 115200);`.

https://stackoverflow.com/questions/3572463/what-is-context-on-android

royseberg, 2018-08-14:
Thank you for your quick reply. That fixes it. It works now at 115200.