omaraflak/Arduino-Library #4: Changing Arduino vendor id

fernando-s97, 2018-03-07:
Hi,

Is it possible to change the vendor id? Mine one has a different value then 9025.

Thank you.

OmarAflak, 2018-03-07:
Hi,

What is the vendor id of your Arduino ?

fernando-s97, 2018-05-09:
0x0403

OmarAflak, 2018-05-09:
I update the library, you should be able to use it now.

`implementation 'me.aflak.libraries:arduino:1.4.2'
`

closed 2018-05-09 by OmarAflak

fernando-s97, 2018-05-18:
Thank you! I'll try the new version.

skgadi, 2018-06-11:
Hi Omar,

I think it is useful to extend this library to support more than just the Arduino vendor IDs.

Is it possible to extend the support to all the vendor IDs?

I sometimes use Microchip USB micro-controllers and they use a different vendor ID.

Thank you.

OmarAflak, 2018-06-12:
Hi @skgadi, it surely is possible.

The library is called Arduino, so it wouldn't be very consistent if I opened the lib the all vendor ids.

That said, you can clone the repo and make your own library. It's only one class, it's very easy.

skgadi, 2018-06-13:
Thank you. I will do it :)

Also, I want to bring to your notice that with the new update of 1.4.2 my phone is not recognizing arduino uno board.

OmarAflak, 2018-07-22:
@skgadi As you know from the other issue, I update the library to version **1.4.3**.
I added a `Log.i(TAG, vendorId)` in the library code, which means that if you take a look at your logcat when wiring your board you will see its vendor id printed.
Once you got it,  simply use `arduino.addVendorId(vendorId)` and it should be fine.

skgadi, 2018-07-22:
@OmarAflak , Thank you for the update. It is working as expected.

Also, thank you for adding support for new vendor IDs.