omaraflak/Arduino-Library #10: Phone crashes

davidkrammer, 2019-05-01:
My Smartphone (Samsung Galaxy S10+) crashes after around a minute using your sample Code.
It simply stops working and shutsdown and restarts.

Before it shutsdown the code works really fine! Everything works as expected.

I dont have any Logs, as i cannot connect my PC with Android Studio and the Arduino (Wemos D1 mini in my Case) at the same time..

Any ideas?

OmarAflak, 2019-05-04:
Hey @davidkrammer, thank you for the feedback !

There is a way to get the logs. Do you think you could connect your phone to your pc using adb over TCP ?

Steps to do it :

1. If your phone doesn't have this option in the developer section, you can download this app : https://play.google.com/store/apps/details?id=com.ttxapps.wifiadb&hl=en
2. Start ADB over TCP. Your phone and your pc need to be on the same network
3. In Android Studio's terminal, connect to your phone using `adb connect <ip_address>` on your pc

You should be able to see the logs now :)

Keep me updated !

davidkrammer, 2019-05-14:
When the phone crashes the Logs from the wifiadb disappear also. But i found a temporary fix. 
The only issue with this is, that i need to restart my Application if i disconnect and reconnect the Arduino. Else it won't get recognized by my App.

OmarAflak, 2019-05-14:
I'm glad you found a solution for your case, though it would nice to know why it crashed in the first place. The default behavior of Android Studio is to show the logs of the "selected application". When it crashes, the application is no longer selected and the logs disappear. You can change that in your settings. You just need to select `Edit Filter Configuration…` and create a filter for your app specifically.

Thanks !

OmarAflak, 2019-05-14:
Also, check out this sample code. It may help you in case you're using the lib in the wrong  way :

https://github.com/OmarAflak/Arduino-Library/blob/master/app/src/main/java/me/aflak/libraries/MainActivity.java