omaraflak/Arduino-Library #12: About the receiving data from arduino to android

JMJ15, 2019-11-23:
1st of all, thanks for your codes. I need to transfer and receive data between arduino and android. Send data from ardroid to arduino is okay but reverse is not working. 
@Override // arduino sent a message
    public void onArduinoMessage(byte[] bytes) {

        display(" Working! this function");
        display("> " +new String(bytes));
    }
This part is not working
rest all are fine.
I am very new to android. I am not getting idea to debug it. Seeking help

OmarAflak, 2019-11-23:
Hi @JMJ15, thanks for reaching out.
How are you sending information from Arduino ? The library will display any text ending with "\n". Basically, you need to call `Serial.println("blabla")` from Arduino.
Let me know if it worked !

JMJ15, 2019-11-23:
In readme  (for arduino side that Example code which sends back every character received) its given as Serial.print(). So i  try that, I try serial.write() but it not worked. Thanks for your help...When i try with serial.println() even with serial.write() it works..

JMJ15, 2019-11-23:
One more doubt. When I try to add another vendor Id as arduino.addVendorId(XXXX); in android program, then product(Arduino 101 and Arduino Due) with that vendor ID did not get connected with android. I didn't get the message to accept the device in android program.

OmarAflak, 2019-11-23:
Great! Right, I forgot to update the Readme.
Are you sure about the vendor id you're entering ? If it's an official Arduino the vendor id should be the one in the library.

JMJ15, 2019-11-23:
yeah I am sure about the vendor id. Because i check it with board info in arduino and also through control panel via taking details of connected device.  It will be a 4 digit number right? Like 8087 or 2341(arduino uno)..Even if im checking the board info it shows like 0x2341 for arduino. Only need to take the last 4 digit?

OmarAflak, 2019-12-05:
This issue was resolved in https://github.com/OmarAflak/Arduino-Library/issues/13.

closed 2019-12-05 by OmarAflak