omaraflak/Arduino-Library #13: For changing Vendor ID

JMJ15, 2019-11-25:
Hey,
I tried this code with arduino Uno (Vendor ID :0x2341) the code is working fine. But when I connect the product with Vendor ID: 1B4F and product ID: 9D0F for same purpose but it didn't worked with the code
arduino = new Arduino(this);
       //arduino.addVendorId(1B4F);
The product is a SAMD21 microprocessor with CDC USB profile. Is it not possible to do with this?

vlaminck, 2019-11-26:
I have the same problem. I have an ESP32 which has a vendorId of `0x10c4`. Is there any way to connect to it using this library?

JMJ15, 2019-11-27:
Is it working fine when u try with arduino.addVendorId(0x10C4) ?

JMJ15, 2019-11-27:
Thanks a lot both of you. it works when I try with this code arduino.addVendorId(0x1B4F);

closed 2019-11-27 by JMJ15