omaraflak/Arduino-Library #16: App wont function on phone

Reinkai, 2020-06-16:
Hi, i used your code to create an android app for arduino. Then i build the apk and download on my phone. But the app doesnt function and phone supply power to arduino only. Should i open usb debugging when i use the app. 

Sorry i am new to android studio.

Thanks.

OmarAflak, 2020-06-16:
Hey. You need to enable OTG in your phone settings.

Reinkai, 2020-06-16:
The OTG is automatically enabled and usb is controlled by arduino. is it because my apk is              
 app-debug.apk.

Thanks.

mepix, 2020-07-25:
@Reinkai, I have the current version of the library working on an Android with USB-C. OTG seems to be enabled automatically on the newer phones with USB-C. I find another common issue is when using non-genuine Arduinos that have a different vendor ID from the default. Since the library filters by vendor ID, you will need to add other IDs when setting up the library in your code.

meet2011, 2020-08-18:
@OmarAflak 

I'm using samsung m series phone, I don't see anything related to OTP in settings. This code doesn't work.