omaraflak/Arduino-Library #21: Arduino data not showing on Android App

cosminciorba, 2021-03-01:
Hi, it works sending data to the Arduino but when i want to show the data from the Arduino on a (textView for example) just doesn't show anything. I would appreciate your help :)

OmarAflak, 2021-03-01:
You need to add a `\n` to the end of the message when sending it from the Arduino.

cosminciorba, 2021-03-01:
[image]
[image]
This is my Arduino and Java code for sending /reciving, still not showing anything, maybe i'm doing something wrong

OmarAflak, 2021-03-01:
First, on Arduino use `Serial.println` instead of `Serial.print`. Also you should read a whole line before sending it. Right now you send every character separately, and when you receive the character on Android you overwrite the TextView content with it. Log the results on Android to make sure you are receiving the characters.